piersfinlayson/otb-iot #1: First approach user experience

PieBru, 2016-12-30:
Hi Piers,
I totally embrace your otb-iot philosophy, particularly about robustness and the choice to base it on open-esp-sdk.
I'm building some critical IoT devices for my home (fllooding, well level, ...), so I'm trying otb-iot on my Arch Linux dev pc. Here is my first feedback.
1. open-esp-sdk installation doc refers only to Debian, in Arch I installed these prerequisites without checking if someone could be skipped:
`$ yaourt --needed -S make unrar autoconf automake libtool gcc gperf flex bison texinfo gawk ncurses python sed git unzip bash help2man wget bzip2 libtool`
2. To solve otb-iot build errors, I did some patches to the Makefile:
```
MQTT_CFLAGS = -Ilib/mqtt -Ilib/httpd **-Ilib/brzo_i2c** -std=c99
ESPTOOL_PY = **python2** $(XTENSA_DIR)/esptool.py
```
3. As I didn't use PlatformIO, to get the serial console I used miniterm2.py:
`$ miniterm2.py /dev/ttyUSB1 115200`
4. After `flash_initial`, the serial console shows only binary data even with the speed correctly set to 115200.
5. After booting the ESP doesn't activate any AP.
otb-iot seems very promising to me.
Thank you, cheers from Italy.
Piero

piersfinlayson, 2016-12-30:
Hi Piero - after flashing try connecting at 230400 or 74880, and see if you get more useful output.  Once you get useful serial output this should make it clear what the issue preventing the AP being brought up is.

PieBru, 2016-12-30:
Thanks Piers, I actually tryed all speeds but 74880 :(
A problem seems signaled by *rf_cal[0] !=0x05,is 0xFF*
I found the message in the elf disassembly, but not inside .c or .h
Now I'm using a Wemos D1 mini V1, waiting for some Pro in the mail.

Here is the serial console output, this sequence loops forever:

BOOT: OTA-BOOT v0.1
BOOT: OTA-Boot based on rBoot v1.2.1 - https://github.com/raburton/rboot
BOOT: Checking GPIO14 o
BOOT: Flash Size:   32 Mbit
BOOT: Flash Mode:   QIO
BOOT: Flash Speed:  40 MHz
BOOT: Option: Big (>1MB) flash
BOOT: Option: Config checksum
BOOT: Option: IROM checksum
BOOT: Booting rom 0 at 0x0004bbf0
BOOT: mmap 0,0,1
rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 2156, room 16
tail 12
chksum 0x4f
ho 0 tail 12 room 4
load 0x3ffe8000, len 1104, room 12
tail 4
chksum 0x3c
csum 0x3c

Thank you,
Piero

piersfinlayson, 2016-12-30:
The problem is I’ve been a bit lazy and am not flashing the proper version of esp_init_data_default.bin onto the device.  If you look in the Makefile you can see I’m creating the file ff.bin and then flashing it to 0x3fc000 on the device.  The version of the SDK you’re using is then objecting to this.

The quickest fix would be to change the Makefile to instead of flashing bin/ff.bin to flash the real esp_init_data_default.bin.  On my install that’s at:

/opt/esp-open-sdk/ESP8266_NONOS_SDK_V1.5.4_16_05_20/bin/esp_init_data_default.bin

Give this a try and let me know how you get on.  I’ll give it a go myself when I have time.

Would be interested to know what version of the SDK you’re using - as I’ve never seen this problem (despite having d1 minis all over the place!).

piersfinlayson, 2016-12-30:
Have updated the Makefile will this change, plus also incorporated fixes for #1/2 above.

Commit:
https://github.com/piersfinlayson/otb-iot/commit/fb73ec67e9c6225ce877e6011fea61a2390cbd32

PieBru, 2016-12-30:
> fb73ec6

Thank you, ESP booted, activated the AP and replys to icmp ping.

However the page at http://192.168.4.1 is not served, the browser timeouts.
I tryed using firefox and chromium, then verifyed with tcpdump: the server
does not reply data to GET / HTTP/1.0
Confirmed manually using netcat.

To skip this step I tryed to force client mode by using the code under #if
0 into otb_main.c@53, but the function otb_wifi_set_stored_conf() seems
missing from github sources.
I succeeded connecting to mosquitto on my pc by placing this code into
otb_main.c after after the call to otb_conf_load():
//#if 0
  // Force an SSID/password/MQTT
  wifi_set_opmode_current(STATION_MODE);
  extern otb_conf_struct otb_conf_private;
  otb_conf_struct *conf = &otb_conf_private;
  strcpy(conf->ssid, "myssid");  // See OTB_CONF_WIFI_SSID_MAX_LEN
  strcpy(conf->password, "mypassword");  // See
OTB_CONF_WIFI_PASSWORD_MAX_LEN
  strcpy(conf->mqtt.svr, "192.168.1.182");
  INFO("Pausing 20s ...");
  otb_util_delay_ms(20000);
//#endif

So mosquitto server confirms:
New client connected from 192.168.1.186 as OTB-IOT.02f422 (c1, k120).

Thanks,
Piero

PieBru, 2016-12-30:
>> Would be interested to know what version of the SDK you’re using - as I’ve never seen this problem (despite having d1 minis all over the place!).

I cloned the sdk this morning at 10:04 CET, the sdk directory links to ESP8266_NONOS_SDK_V2.0.0_16_08_10
```
[esp-open-sdk]$ date
Fri Dec 30 23:40:42 CET 2016
[esp-open-sdk]$ git status
On branch master
Your branch is up-to-date with 'origin/master'.
```

piersfinlayson, 2017-01-01:
> However the page at http://192.168.4.1 is not served, the browser timeouts.
I tryed using firefox and chromium, then verifyed with tcpdump: the server
does not reply data to GET / HTTP/1.0

This is a bit odd.  I've just tested with a reset device here and when I connect to the AP the captive portal successfully redirects my browser to http://192.168.4.1/.  And I can also manually browse to it (this is using chrome).

Suggests to me that something else more basic isn't working - for example you don't have DHCP enabled on the device you're using to connect to it, or for some reason your device isn't getting a DHCP lease.

PieBru, 2017-01-02:
Solved.
I built from a freshly cloned repo and did worked!

To understand what happened I renamed my local otb-iot repo and got a new clone, exactly as I initially did two days ago. The only diff from src/ and include/ between my repo and the fresh one was the code I added under `#if 0` in otb_main.c (see my comment above), so I disabled itand rebuilt. It still didn't work.
I tryed also uncommenting and then commenting the MQTT_HOST in mqtt_user_config.h, no benefits.

Without modifying any source, I then restarted from `make clean` and finally my local repo worked. I didn't investigate further as I'm happy it isn't a source code issue and the solution is silly simple. IMHO it coud be caused by a residual state from the ff.bin initial build failures.

Now my regular build command line is a few seconds slower, but more failsafe:
`$ **make clean** && make && make flash_initial && miniterm.py /dev/ttyUSB0 115200`

Thank you,
Piero

closed 2017-01-02 by PieBru

piersfinlayson, 2017-01-02:
Glad it's working!