piersfinlayson/otb-iot #3: Basic manual testing #2: mqtt commands and replys

PieBru, 2016-12-31:
I'm new to mqtt. Before writing some code to exploit its functionality I want to manually try and understand its basic funcionality. I'm using my Arch Linux PC.
I'm silly stuck, once solved this doc could be part of a basic testing guide for otb-iot newcomers.

First I installed mosquitto on my PC and spawned a verbose broker in a dedicated terminal window:
```
$ yaourt -S mosquitto
$ mosquitto -v
```
On another dedicated PC terminal window I started a subsciber to the otb-iot main topic:
`mosquitto_sub -v -t /otb-iot/02f422/status`

By sending a message to the status topic on the broker:
```
$ mosquitto_pub -d -t /otb-iot/02f422/status -m ping
Client mosqpub/28211 sending CONNECT
Client mosqpub/28211 received CONNACK
Client mosqpub/28211 sending PUBLISH (d0, q0, r0, m1, '/otb-iot/02f422/status', ... (4 bytes))
Client mosqpub/28211 sending DISCONNECT
```
 it is shown correctly by the PC subscriber, so the broker seems to work:
`/otb-iot/02f422/status ping`

Then I configured otb-iot on my Wemos to connect to the mqtt broker on my PC, waiting for a solution to the HTTP problem I used the trick explained in issue #1.
After pressing the reset button on the Wemos, it initialized mqtt and then connected to the mosquitto broker with otb-iot subscriptions. I don't see /otb-iot/.../system here, is it OK?
```
...
MQTT: Connected to broker 192.168.1.182:1883
MQTT: Connected to 192.168.1.182:1883
MQTT: Subscribe: /otb-iot/02f422
MQTT: Subscribe: /otb-iot/all
MQTT: Subscribe: /otb-iot/all
MQTT: Subscribe: /otb-iot/02f422
...
```
My PC subscriber showed the Wemos boot message, good sign:
`/otb-iot/02f422/status booted:slot:0`

and the broker console shows periodic Wemos keepalive ping, another good sign:
```
1483183790: Received PINGREQ from OTB-IOT.02f422
1483183790: Sending PINGRESP to OTB-IOT.02f422
```

At this point would like to try some manual request/reply, but I'm missing something about mqtt and issuing
`$ mosquitto_pub -d -t /otb-iot/02f422/system -m ping`
or other info requests I did not receive any reply. Actually it seems the Wemos never got the message (DEBUG is enabled), so probably the request doesn't leave the broker. I unsuccessfully tryed all topic variations that came in my mind.

Any hint?
Happy new year,
Piero

piersfinlayson, 2016-12-31:
I’ve changed the format of MQTT commands recently.  This should work:

mosquitto_pub -t /otb-iot/02f422 -m trigger/ping

Take a look at the big comment in include/otb_cmd.h for the complete list of commands (although some aren’t supported yet).

PieBru, 2016-12-31:
Solved.
I fully share your way to think tools architecture.
Thank you,
Piero

closed 2016-12-31 by PieBru