piersfinlayson/otb-iot #41: `.text' will not fit in region `iram1_0_seg

shouidar, 2018-09-25:
I try to compile otb-iot on a freshely set up Virtual machine with open-sdk (latest)
after correcting some conflicting declarations in esp_systemapi.h (commented as they seem to be already declared in the last open-sdk osapi.h)
I'm getting the following error:
ld: bin/app_image.elf section `.text' will not fit in region `iram1_0_seg'

Any suggestions where to start looking

thanks

piersfinlayson, 2018-09-25:
Sigh. This this will be the latest sdk building more stuff that lives in
ram into the image. I'll probably have to shuffle more of otbiot onto flash
which will be a bit fiddly.

I'm travelling right now so may not be able to look at until the weekend.

piersfinlayson, 2018-09-29:
Actually, it builds fine with 2.1.0, and has a .text size of 0x7b80.  2.1.0-18-g61248df is the version which is failing.

piersfinlayson, 2018-09-29:
Here's the functions which are being linked into IRAM with 2.1.0-18-xxx, and which are in IROM with 2.1.0:

wDev_Get_KeyEntry
ieee80211_rate_ref_init
ieee80211_freedom_inside_cb
ieee80211_freedom_init
ieee80211_user_ie_init
ieee80211_ifattach
ieee80211_mhz2ieee
ieee80211_chan2ieee
ieee80211_ieee2mhz
wifi_mode_set
ieee80211_crypto_encap
ieee80211_crypto_decap
ieee80211_getmgtframe
ieee80211_hostap_attach
hostap_handle_timer
hostap_input
freqcal_scan_done
wifi_softap_start
wifi_softap_stop
ieee80211_ht_attach
ieee80211_ht_node_init
ieee80211_parse_htcap
ieee80211_ht_updateparams
ieee80211_setup_htrates
ieee80211_setup_basic_htrates
ieee80211_add_htcap
ieee80211_add_htcap_vendor
ieee80211_add_htinfo
ieee80211_add_htinfo_vendor
ieee80211_deliver_data
ieee80211_decap
ieee80211_setup_rates
ieee80211_alloc_challenge
ieee80211_parse_beacon
ieee80211_parse_wpa
ieee80211_parse_rsn
ieee80211_parse_action
ieee80211_mesh_quick_init
ieee80211_mesh_quick_deinit
ieee80211_mesh_quick_set
ieee80211_mesh_quick_get
ieee80211_mesh_quick_ie_add
ieee80211_output_pbuf
ieee80211_send_setup
ieee80211_mgmt_output
ieee80211_send_nulldata
ieee80211_add_rates
ieee80211_add_xrates
ieee80211_send_probereq
ieee80211_getcapinfo
ieee80211_send_mgmt
ieee80211_alloc_proberesp
ieee80211_send_proberesp
ieee80211_alloc_deauth
ieee80211_send_deauth
ieee80211_freedom_output
ieee80211_beacon_alloc
ieee80211_psq_init
ieee80211_set_tim
ieee80211_pwrsave
pwrsave_flushq
ieee80211_node_pwrsave
ieee80211_proto_attach
ieee80211_iserp_rateset
ieee80211_mlme_connect_bss
ieee80211_regdomain_detach
ieee80211_alloc_countryie
ieee80211_chan_in_regdomain
ieee80211_update_regdomain
ieee80211_rfid_locp_recv_open
ieee80211_rfid_locp_recv_close
ieee80211_rfid_locp_recv_reset
ieee80211_rfid_locp_recv
register_ieee80211_rfid_locp_recv_cb
unregister_ieee80211_rfid_locp_recv_cb
ieee80211_scan_attach
freq_changle_enable_softap_link_check
freq_changle_scan_start
scan_start
scan_pm_channel_op_cb
scan_cancel
scan_add_bssid
scan_remove_bssid
scan_hidden_ssid
scan_add_probe_ssid
scan_remove_probe_ssid
scan_clear_channles
scan_set_desChan
scan_get_type
cannel_scan_connect_state
scan_connect_state
scan_profile_check
scan_parse_beacon
sta_status_set
ieee80211_sta_new_state
sta_input
wifi_station_start
wifi_station_stop
chm_init
chm_acquire_lock
chm_release_lock
chm_start_op
chm_end_op
chm_cancel_op
chm_return_home_channel
chm_set_current_channel
chm_freq2index
chm_check_same_channel
cnx_attach
cnx_sta_connect_led_timer_cb
cnx_sta_connect_cmd
cnx_sta_scan_cmd
cnx_connect_timeout
cnx_auth_timeout
cnx_assoc_timeout
cnx_handshake_timeout
cnx_start_handoff_cb
cnx_csa_fn
cnx_bss_alloc
cnx_rc_search
cnx_add_rc
cnx_remove_all_rc
cnx_remove_rc
cnx_rc_update_rssi
cnx_rc_update_state_metric
cnx_rc_update_age
cnx_update_bss
cnx_update_bss_more
cnx_sta_leave
cnx_sta_associated
cnx_node_alloc
cnx_node_remove
cnx_node_search
cnx_node_leave
wifi_softap_staconnected_event_policy
wifi_softap_toomany_deny
cnx_node_join
ieee80211_send_action_register
ieee80211_send_action_unregister
ieee80211_send_action
ieee80211_recv_action_register
ieee80211_recv_action_unregister
ieee80211_recv_action
ieee80211_avs_fetch_key_idx
ieee80211_avs_clear_key_idx
get_iav_key
register_ieee80211_action_vendor_get_key_cb
unregister_ieee80211_action_vendor_get_key_cb
ieee80211_add_action_vendor_spec_esp
ieee80211_alloc_action_vendor_spec
ieee80211_send_action_vendor_spec
ieee80211_recv_action_vendor_spec
ieee80211_action_vendor_spec_attach
register_ieee80211_action_vendor_spec_cb
unregister_ieee80211_action_vendor_spec_cb
ieee80211_add_ie_vendor_esp_head
ieee80211_add_ie_vendor_esp_mesh_assoc
ieee80211_add_ie_vendor_esp_mesh_group
ieee80211_add_ie_vendor_esp_simple_pair
ieee80211_add_ie_vendor_esp_freq_annon
ieee80211_add_ie_vendor_esp_now
ieee80211_add_ie_vendor_esp_ssid
ieee80211_add_ie_vendor_esp_manufacturer
ieee80211_phy_init
ieee80211_phy_type_get
ieee80211_setup_ratetable
hostapd_config_defaults_bss
hostapd_config_defaults
hostapd_mac_comp
hostapd_mac_comp_empty
hostapd_setup_wpa_psk
hostapd_wep_key_cmp
hostapd_maclist_found
hostapd_get_psk
hex2byte
hexstr2bin
wpa_get_ntp_timestamp
wpa_config_parse_string
dup_binstr
ets_strdup
r_rand
os_random
os_get_random
wpa_init
wpa_auth_sta_init
wpa_auth_sta_associated
wpa_auth_sta_deinit
wpa_receive
__wpa_send_eapol
wpa_remove_ptk
wpa_auth_sm_event
wpa_write_rsn_ie
wpa_auth_gen_wpa_ie
wpa_add_kde
wpa_validate_wpa_ie
wpa_parse_kde_ies
wpa_sm_rx_eapol
wpa_register
wpa_set_profile
wpa_set_pmk
wpa_set_bss
pp_michael_mic_failure
eapol_txcb
wpa_sm_set_state
wpa_parse_wpa_ie_rsn
wpa_parse_wpa_ie_wpa
wpa_eapol_key_mic
wpa_compare_rsn_ie
wpa_pmk_to_ptk
rsn_pmkid
wpa_cipher_to_suite
wpa_snprintf_hex_uppercase
wpa_snprintf_hex
wpa_parse_wpa_ie
wpa_supplicant_parse_ies
wpa_gen_wpa_ie
ppInstallKey
ppGetKey
wpa_config_profile
wpa_config_bss
wpa_config_assoc_ie
dhcp_bind_check
eagle_auth_done
wpa_neg_complete
wpa_attach
wpa_sm_alloc_eapol
aes_wrap
rijndaelEncrypt
aes_encrypt_init
aes_encrypt
aes_encrypt_deinit
rijndaelKeySetupEnc
ieee80211_find_channel
ieee80211_find_channel_byieee
ieee80211_crypto_attach
ieee80211_crypto_available
ieee80211_crypto_setkey
ieee80211_ht_node_cleanup
ieee80211_setup_rateset
ieee80211_tx_mgt_cb
ieee80211_psq_cleanup
ieee80211_set_shortslottime
ieee80211_wme_initparams
ieee80211_wme_updateparams
ieee80211_regdomain_attach
scan_check_hidden
ieee80211_parse_wmeparams
hostapd_rate_found
inc_byte_array
os_get_time
wpa_auth_for_each_sta
wpa_auth_sta_no_wpa
wpa_auth_uses_mfp
wpa_cipher_key_len
wpa_cipher_to_alg
rsn_cipher_put_suites
wpa_cipher_put_suites
eloop_cancel_timeout
eloop_register_timeout
wpa_sm_deauthenticate
wpa_sm_mlme_setprotection
wpa_sm_get_beacon_ie
wpa_sm_disassociate

piersfinlayson, 2018-09-29:
I have a tentative fix on branch issue-41, which brings .text back down to 0x76bc, by forcing libnet80211 and libwpa to link into IROM. 

@shouidar could you give this a go?  I have no way of testing this fix for the next week, and it's entirely possible it'll crash, if there's any functions in either of those two libraries which shouldn't be linked into IROM (e.g. because they need to execute within an interrupt handler).

This fix assumes the SDK is installed to: /home/esp/esp-open-sdk.  If it's not modify ld/eagle.app.v6.ld and change the paths for libnet80211 and libwpa.  I'll make this handle arbitrary paths before I merge the fix to master.

shouidar, 2018-09-30:
Tested you new fix. compiles and runs fine. Did few mqtt testing and all seems to behave normally.
Thank you for looking into this issue.

piersfinlayson, 2018-09-30:
Thanks for letting me know - I'll keep this issue open to test and commit a full fix and merge to master.

piersfinlayson, 2018-09-30:
Actually, the eagle.app.v6.ld has been modified similarly in 2.1.0-18-g61248df, so the correct fix is probably just to use that.

piersfinlayson, 2018-10-03:
Now have a proper fix in branch issue-41.  Will test this weekend and then merge with master.

piersfinlayson, 2018-10-27:
Fixed in master - have checked it builds and runs with SDK 2.1.0-18-g61248df, on both Wemos D1 mini and otb-iot hardware.

closed 2018-10-27 by piersfinlayson