piersfinlayson/otb-iot #42: error compiled

radioham777, 2018-10-26:
vlad@vlad-MS-7529:~/otb-iot$ make all
mkdir -p bin obj/otb obj/httpd obj/rboot obj/rboot obj/rboot obj/mqtt obj/brzo_i2c obj/hwinfo obj/stage obj/softuart obj/libb64 obj/html
/home/vlad/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-objcopy -W Cache_Read_Enable_New /home/vlad/esp-open-sdk/sdk/lib/libmain.a bin/libmain2.a
/home/vlad/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -Os -Iinclude -Iinclude/boards -I/home/vlad/esp-open-sdk/sdk/include -mlongcalls -c -ggdb -Wpointer-arith -Wundef -Wno-address -Wl,-El -fno-inline-functions -nostdlib -mtext-section-literals -DICACHE_FLASH -Werror -D__ets__ -Ilib/rboot -DOTB_IOT_ESP12 -Ilib/esp8266-software-uart/softuart/include -Ilib/httpd -Ilib/mqtt -Ilib/rboot -Ilib/rboot/appcode -Ilib/brzo_i2c -std=c99 -DOTB_IOT_V0_3 -MMD -c src/otb_ds18b20.c -o obj/otb/otb_ds18b20.o
In file included from include/otb.h:51:0,
                 from src/otb_ds18b20.c:37:
include/esp_systemapi.h:30:13: error: conflicting types for 'ets_timer_arm_new'
 extern void ets_timer_arm_new(ETSTimer *ptimer, uint32_t milliseconds, bool repeat_flag, int isMstimer);
             ^
In file included from include/otb.h:43:0,
                 from src/otb_ds18b20.c:37:
/home/vlad/esp-open-sdk/sdk/include/osapi.h:65:6: note: previous declaration of 'ets_timer_arm_new' was here
 void ets_timer_arm_new(os_timer_t *ptimer, uint32_t time, bool repeat_flag, bool ms_flag);
      ^
In file included from include/otb.h:51:0,
                 from src/otb_ds18b20.c:37:
include/esp_systemapi.h:39:13: error: conflicting types for 'ets_delay_us'
 extern void ets_delay_us(uint32_t us);
             ^
In file included from include/otb.h:43:0,
                 from src/otb_ds18b20.c:37:
/home/vlad/esp-open-sdk/sdk/include/osapi.h:33:6: note: previous declaration of 'ets_delay_us' was here
 void ets_delay_us(uint16_t us);
      ^
In file included from include/otb.h:51:0,
                 from src/otb_ds18b20.c:37:
include/esp_systemapi.h:43:13: error: conflicting types for 'ets_isr_attach'
 extern void ets_isr_attach(int intr, void *handler, void *arg);
             ^
In file included from /home/vlad/esp-open-sdk/sdk/include/os_type.h:28:0,
                 from /home/vlad/esp-open-sdk/sdk/include/osapi.h:29,
                 from include/otb.h:43,
                 from src/otb_ds18b20.c:37:
/home/vlad/esp-open-sdk/sdk/include/ets_sys.h:67:6: note: previous declaration of 'ets_isr_attach' was here
 void ets_isr_attach(int i, ets_isr_t func, void *arg);
      ^
In file included from include/otb.h:51:0,
                 from src/otb_ds18b20.c:37:
include/esp_systemapi.h:49:13: error: conflicting types for 'ets_install_putc1'
 extern void ets_install_putc1(void *routine);
             ^
In file included from include/otb.h:43:0,
                 from src/otb_ds18b20.c:37:
/home/vlad/esp-open-sdk/sdk/include/osapi.h:34:6: note: previous declaration of 'ets_install_putc1' was here
 void ets_install_putc1(void (*p)(char c));
      ^
In file included from include/otb.h:51:0,
                 from src/otb_ds18b20.c:37:
include/esp_systemapi.h:57:15: error: conflicting types for 'ets_strlen'
 extern size_t ets_strlen(const char *s);
               ^
In file included from include/otb.h:43:0,
                 from src/otb_ds18b20.c:37:
/home/vlad/esp-open-sdk/sdk/include/osapi.h:47:5: note: previous declaration of 'ets_strlen' was here
 int ets_strlen(const char *s);
     ^
In file included from include/otb.h:51:0,
                 from src/otb_ds18b20.c:37:
include/esp_systemapi.h:60:12: error: conflicting types for 'ets_strncmp'
 extern int ets_strncmp(const char *s1, const char *s2, int len);
            ^
In file included from include/otb.h:43:0,
                 from src/otb_ds18b20.c:37:
/home/vlad/esp-open-sdk/sdk/include/osapi.h:48:5: note: previous declaration of 'ets_strncmp' was here
 int ets_strncmp(const char *s1, const char *s2, unsigned int n);
     ^
In file included from include/otb.h:51:0,
                 from src/otb_ds18b20.c:37:
include/esp_systemapi.h:74:13: error: conflicting types for 'uart_div_modify'
 extern void uart_div_modify(int no, unsigned int freq);
             ^
In file included from include/otb.h:49:0,
                 from src/otb_ds18b20.c:37:
/home/vlad/esp-open-sdk/sdk/include/user_interface.h:652:6: note: previous declaration of 'uart_div_modify' was here
 void uart_div_modify(uint8 uart_no, uint32 DivLatchValue);
      ^
Makefile:299: recipe for target 'obj/otb/otb_ds18b20.o' failed
make: *** [obj/otb/otb_ds18b20.o] Error 1
vlad@vlad-MS-7529:~/otb-iot$

piersfinlayson, 2018-10-26:
This is essentially related to #41 - from some recent version on rwards the SDK started including function prototypes for the fns I had otbiot had its own (conflicting) definitions for.  It's a trivial fix to get it to compile, but it'll still fail the link step because of the problems in #41.  I'm still figuring out how to fix that properly - taking the new linker file from the SDK seems to leave a non booting ESP :-(.

piersfinlayson, 2018-10-27:
Fixed in master - have checked it builds and runs with SDK 2.1.0-18-g61248df, on both Wemos D1 mini and otb-iot hardware.

closed 2018-10-27 by piersfinlayson