piersfinlayson/otb-iot #46: Problems enabling the serial port

David-Verez, 2019-02-27:
Hi super impresive work. Thanks so much.
I am trying to use otb-iot to obtain the reading of a thermal energy meter through the M-BUS protocol. I am using a Raspberry pi with Mosquitto and a nodeMCU as an MQTT client. I followed all the steps and flash the nodeMCU and managed to successfully connect to the broker in the raspberry. I follow the steps described and disable the serial port, configure the Tx and Rx terminals and the baudrate, but when the serial communication is enabled, the nodeMCU is restarted, the changes are lost and the serial communication is not enabled.
thanks in advance david

This is the error
ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 8868, room 16 
tail 4
chksum 0xb3
load 0x3ffe826c, len 5204, room 4 
tail 0
chksum 0xd4
load 0x3ffe96c0, len 6680, room 8 
tail 0
chksum 0x88
load 0x3ffe8000, len 620, room 8 
tail 4
chksum 0x64
csum 0x64

BOOT: OTA-BOOT v0.4a
EEPROM: Trying internal bus
24XXYY: Failed: 4
EEPROM: Failed to connect to eeprom at 0x57 0
EEPROM: Failed
BOOT: Checking GPIO14 o
BOOT: Flash Size:   32 Mbit
BOOT: Flash Mode:   DIO
BOOT: Flash Speed:  40 MHz
BOOT: Option: Big (>1MB) flash
BOOT: Option: Config checksum
BOOT: Option: IROM checksum
BOOT: Booting rom 0 at 0x000532b0
BOOT: mmap 0,0,1
⸮oo⸮⸮oo⸮⸮|⸮⸮l⸮⸮⸮⸮⸮oN⸮

piersfinlayson, 2019-02-28:
I've not seen this - but then I haven't tried this function on the nodemcu hardware.  Can you provide each of the MQTT commands you're sending before the restart?

Also, what SDK version are you using to build otb-iot?

I think I have a nodemcu somewhere so can try to reproduce the problem.

David-Verez, 2019-02-28:
I get the above error message from a serial monitor in window.
I'm using the Quick-Start flashing method to build otb-iot in the nodeMCU. 

The exact MQTT commands that I send are
mosquitto_pub -t /otb-iot/3c71bf291eaa -m set/config/serial/disable
mosquitto_pub -t /otb-iot/3c71bf291eaa -m set/config/serial/tx/15
mosquitto_pub -t /otb-iot/3c71bf291eaa -m set/config/serial/rx/13
mosquitto_pub -t /otb-iot/3c71bf291eaa -m set/config/serial/speed/2400
mosquitto_pub -t /otb-iot/3c71bf291eaa -m set/config/serial/enable

piersfinlayson, 2019-02-28:
I think I spot the bug with a quick code read:

In otb_serial.c line 568:

`
          if (((conf->mezz_info == NULL) && ((conf->rx == OTB_SERIAL_PIN_INVALID) || (conf->tx == OTB_SERIAL_PIN_INVALID))) ||
              ((conf->mezz_info == NULL) && (!conf->mezz_info->use_mezz)))
`
The second line (conf->mezz_info == NULL) should be != NULL.  It's attempting to use derefence a null pointer when using non-otb-iot hardware (like a nodemcu).

Fix is trivial, rebuilding the quick-start may have to wait til the weekend.

David-Verez, 2019-02-28:
Thanks, I will try to fix it manually.

piersfinlayson, 2019-03-01:
I've built a new version of the quickstart container image for x86_64.  Instead of using the piersfinlayson/otbiot container use piersfinlayson/otbiot-amd64:0.1.3

Note - I haven't tested this image at all, so this or another problem may still be present!

I'll build and test a combined x86_64/arm image over the weekend.

piersfinlayson, 2019-03-02:
I've verified the problem and fix on a WeMoS D1 mini.  The quickstart container image is now updated.  To get the new one run:

`
docker pull piersfinlayson/otbiot
`

Then run the container again.  Rather than doing a flash_initial, you should be able to get away with a flash_app, which will retain the wifi/MQTT credentials:

`
make flash_app && make con
`

Let me know how you get on.  In particular if you do manage to get this working I'd be interested to find out what make/model of heat meter you're using, and what the mbus output of the device looks like.

closed 2019-03-02 by piersfinlayson

David-Verez, 2019-03-02:
Hello Piers it works !!! activates serial communication without problems. On Monday I will test if I can establish communication with the thermal energy meter, read the parameters and decode them with the libmbus library.
I am using two zenner zelcius c5 isf as an energy meter to obtain the heating and hot water consumption parameters to be added to influxdb and later to visualize it in Grafana together with other sensors that I have installed.
As soon as I have the first output of the mbus I send it to you.

thanks David

piersfinlayson, 2019-03-02:
Great news!

David-Verez, 2019-03-13:
hi Piers I have not sent you the MBUS output because I have not managed to get a response yet from the two zenner energy meters. After the modifications made to the ESP the serial port is enabled but when sending trigger / serial / send / 105b015c16
and trigger / serial / buffer / dump
I only get ok xx

Already check and the counters are in the position addr 1 and 2. Any ideas about it.

piersfinlayson, 2019-03-13:
There's lots of things it could be.  What do you have between the ESP serial pins and the meters?

David-Verez, 2019-03-13:
I have the circuit from otb-iot.readthedocs.io

piersfinlayson, 2019-03-13:
Do you have an oscilloscope or logic level analyser, to check the output either of the serial port from the nodemcu, or the activity on the M-Bus?

David-Verez, 2019-03-13:
no, I do not have. next week I will get a mbus-usb converter that will use
it together with libmbus in a raspberry to see if it detects the counter.

piersfinlayson, 2019-03-13:
That's a good plan.

Other things you can try are;
- changing the baudrate (although the manual for the Zenner does seem to suggest it is 2400 baud)
- trying on other addresses - for example sending 105b025c17 (I think) for address 2.  You can use libmbus command mbus-serial-scan in debug mode to get the appropriate hex commands to query other addresses, as you need to update the checksum as well as address.  You can also see what sequence it uses to reset the bus first, which might help.

David-Verez, 2019-03-13:
great idea, thanks. as soon as I have some news I let you know.